PNKP237/EHU32 #3: Additional information without electric climate

Vortecsz, 2024-05-04:
Hello, would be possible to display coolant, speed, rpm and voltage without the Electronic climate? I know how to activate the electronic climate tab in the bottom but I don't have the automatic climate so this is useless for me. Would be possible to show maybe the coolant and battery in the first page of the screen without auto climate?

PNKP237, 2024-05-04:
Battery is out of the question for now, I haven't been able to come up with a way to read battery voltage from MS-CAN other than through ECC.
As for coolant temp and vehicle speed, it is possible to read it using measurement blocks from the display (but then there's no RPMs nor voltage). I'll try to implement that in the next version (it will try reading reduced data from display if there is no response from ECC) - I guess you can expect the new version in one or two weeks, still optimizing and testing stuff.
As for why - it uses the same protocol that OPCOM or Tech2 use for reading measurement data and only ECC offers coolant temp, speed, rpms and voltage combined. If these tools can't access more data from display or radio neither can we.

Haven't had much luck with sniffing the coolant, rpm and speed data like it was defined [HERE](https://github.com/JJToB/Car-CAN-Message-DB/blob/master/Opel/Astra/H/MS-CAN/Body.md). In my case (Z16XER) I was reading total garbage, but once I get reading basic data from the display done I'll try again with these.

Vortecsz, 2024-05-04:
Love to hear that, it would be possible to display like the coolant in the first page? I would like to have the coolant temperature in the main page without need to swap page everytime

PNKP237, 2024-05-05:
First page as in the mode where song data is shown? I guess I can make it like this but it would require me to rewrite a lot of current logic.
I would much rather show the coolant temp in a "notification box" that sometimes appears in the top right corner of GID/CID, able to be toggled on or off. Do you think that would work for you?

Vortecsz, 2024-05-05:
[image]
That's my current setup, in the top the current media, like aux etc in the bottom consumption etc, my main goal would be to add like coolant in the top right, but if is it hard, I would understand maybe one day I will understand how to program that haha. The notification box would be also not bad at all

PNKP237, 2024-05-05:
Alright, you have a single line GID so writing the coolant info to the album or artist fields won't work because you won't see them anyway. I had the single line version as well but I enabled UHP just to make it display all the three lines of text.
The notification box will work fine but it will obstruct the top right part of the display.

As for GID firmware, EEPROM only holds basic data such as settings and carpass (if paired with the radio). and the GID firmware itself is held inside this TSOP flash package I think, difficult to read that data without a TSOP adapter. Don't feel the need to modify the GID firmware anyway, I'd like this project to work plug-and-play with no modification to the rest of the system

[image]

Vortecsz, 2024-05-05:
Oh okay, would be possible to print in the top right with the single line gid? Or just you can't because u need to overwrite an existing text? Like artist aux etc. Btw with the the EEPROM firmware you can change different version 3 line version, single etc, but if you don't want to modify any firmware that's pretty useless. 
I don't know if you know but a Russian guy seel an obd device probably something similar to your setup, that can show DPF status etc, that's only an idea for the project, I don't need DPF status etc. My main focus is to see the coolant temperature and maybe battery in the main page as stock as possible. Because Opel decided to not add any temperature gauge in the cluster...
Thank you for the help.

PNKP237, 2024-05-05:
Yes the notification box can be displayed even with single line GID.
As I have already stated in the [other issue](https://github.com/PNKP237/EHU32/issues/1) there is no way to read DPF status from MS-CAN and that's the only CAN bus I'm going to support in this project, mainly because ESP32 only has a single CAN peripheral. Otherwise I would need additional hardware to be able request DPF measurement blocks from ECU through HS-CAN.
I recently learned that such module exists but making menu interfaces is not something I would like to focus on at this time even though I have already reverse engineered how the menu can be created.
The problem is that if this is put in a vehicle equipped with UHP it could cause problems because that module "fakes" being an UHP interface menu

PNKP237, 2024-05-05:
Nevermind, the notification box does not show up when the main audio page is displayed. I guess I'll try to fit coolant info on the right of the middle line or find another way to display notifications
I'm surprised you prefer the single line version though

Vortecsz, 2024-05-05:
Oh okay, I prefer the single line because I watch a lot the consumption and the autonomy of the car so I would need to swap page every time. The first page of the three line only display audio information and I only use aux so I would of see aux and nothing more. I prefer to see the consumption

PNKP237, 2024-05-05:
The display is not very good at handling longer strings and as a result the coolant temp is printed on top of the title text. Sometimes it's ignored and not printed at all, for very long song titles

[image]
[image]
[image]

I don't know if I should go ahead with this, the result is kinda bad

Vortecsz, 2024-05-05:
You are right, that's not good at all for posting it, I don't know if you can put the string down the song or you can just write in the line of artist/song name. Btw can you give me the code? I can use the coolant on the right of the AUX string? Because I just use aux so is not a problem for me and thank you for all the help and the amazing work you are putting in the project.

PNKP237, 2024-05-05:
You can't "add" anything to the AUX string sent by the radio, you can only overwrite it.
I built this test version on top of EHU32 which is still not ready for release, but once it's finished I'll give you the code with this functionality added, with a way to customize the text displayed to your liking

Tomi8609, 2024-05-06:
I will keep an eye on this part of the project!

PNKP237, 2024-05-06:
Just pushed v0.9, added an additional mode (accessible by long pressing "3") for displaying just the coolant temp in a single line, looking like so:
[image]

Vortecsz, 2024-05-07:
That's beautiful thank you so much for the work! Currently waiting for the esp and the modules to arrive to get it working.

Wop57, 2024-05-12:
There is an example for showing information in the electronic climate tab. It is in the issue #3 in the Car-CAN-Message-DB of JJToB. Here is the link: 
https://github.com/JJToB/Car-CAN-Message-DB/issues/3

PNKP237, 2024-05-12:
> There is an example for showing information in the electronic climate tab. It is in the issue #3 in the Car-CAN-Message-DB of JJToB. Here is the link: [JJToB/Car-CAN-Message-DB#3](https://github.com/JJToB/Car-CAN-Message-DB/issues/3)

Thanks, never noticed that one.
Even though I have considered showing the temperatures there it still does not solve the underlying issue - if the car does not have ECC (or at least not enabled in Display Variant Programming, changing which requires opcom, which kinda defeats the "as cheap and easy as possible" approach) then we can't utilize that field. 

Already tested with UHP but it does not send any notification boxes, only menu-like screens.
I'll do some additional research to see if I can simulate Navi notifications, but that has to wait since I'm super busy with other things. Also I need to get the CD ROM drive back into my Navi since it refuses to display anything without the drive mechanism.

Vortecsz, 2024-05-15:
Hello is possible to change the pinout with this board?
CP2102 Type-C 38P

[image]

PNKP237, 2024-05-16:
I don't exactly understand what do you mean by "change the pinout", please elaborate.
Is it just changing the pins, as in you'd prefer the CAN TX/RX or I2S on different pins than provided in my example?

Vortecsz, 2024-05-16:
Oh yes sorry, I searched for the defaults pin out of can Rx tx and i2s, example for can Rx and tx io4 and io5  wired them up in my esp board but isn't working at all. Maybe is the board? And I need to define manually the gpio? I find that the default pins are the same for the 38 pins board and the 30 pins board so I don't know why isn't working

Vortecsz, 2024-05-16:
After reflashing the esp every time I connect the board to the car the aux text go away and the current date is displayed is like the radio doesn't work because the buttons don't do nothing but I can hear the sound from the speaker so the radio is on. If I remove the esp everything work as usual

PNKP237, 2024-05-16:
Curious. I do most development on a setup like yours (GID and I assume CD30MP3) and never had the radio stop responding. What model year is your vehicle?

PNKP237, 2024-05-16:
i2s is just an output and if you leave it not connected it does not matter to the ESP32 at all (it doesn't receive any data from PCM5102).

EHU32 will enable bluetooth and start sending stuff to the screen after it notices that the radio has sent some text to the display, it starts with the "1" mode active (audio metadata). You don't have to connect to switch to the "3" mode (1-line coolant info).

If you can, please download version 0.7 from releases and try connecting your phone and playing something to see if the audio title is displayed at all. Of course it won't have the "3" mode and "2" won't work but I would like to see if the "radio message blocking" feature is the problem here (v0.7 is the last version without it).

Vortecsz, 2024-05-16:
Okay, my problem is the 5v and grounds using a car charger with USB port. Gonna try others configuration. Anyway mode 1 work with song names, mode 2 show coolant hope I'm correct, mode 3 show speed and rpm, the rpm and speed are wrong when I'm idle at 800 it shows 3000 same for the speed I'm going 90kmh and it shows 45. Mode 2 coolant is partially wrong if I long press settings and after bc I can access the secret menu of the gid and I can see the little difference between the two coolant value, your is a bit high like 3/4 °c. The audio quality sucks, but that's my problem with the connection, I would love to see you PCB design did you manage to get that done?

PNKP237, 2024-05-16:
It's the other way around, mode 2 should show speed and RPMs and if there is no ECC detected it will try to pull that data from CAN but as I said earlier in my vehicle I would read random values so I haven't had much luck testing that. I guess I'll remove that option from next version since it doesn't work that well - I'll make it only show speed that's read from the display.

I think that the ground of the audio input in CD40 might be either floating or coupled with caps to ground, I'd suggest to measure with a multimeter between the audio input ground and vehicles ground to see if there is any voltage there. If there is then one fix would be to either modify the radio or buy some kind of an audio separation transformer.

Thanks for the feedback, I'll correct the speed display (just divide it by 2 I guess?) when I have some free time next week to possibly test it within my vehicle. As for coolant I'd say it's the GID that's lying since these cars notoriously run hot.
As for the PCB... I'll get around to finish it one day but I've got a lot of things going on right now.

Vortecsz, 2024-05-16:
Okay I resolved all the problem with the connections, the problem is that the audio quality sucks really bad. I don't know why the power is from a car charger with a USB cable, not from the radio.

PNKP237, 2024-05-17:
I hope your PCM5102 isn't broken, here's a test program for you to flash the ESP32 with, please connect it to an amplifier at home so you can check whether the audio output works properly. It has no CAN functionality so you can connect to it and just play music from your phone.
[ESP32_Audio_Stereo.zip](https://github.com/PNKP237/EHU32/files/15345653/ESP32_Audio_Stereo.zip)

Vortecsz, 2024-05-19:
As soon as I have a chance I'll try it thank you very much. Just a question: the ESP power supply should be given through the USB connector or i can give it directly to the vin and gnd pins?

PNKP237, 2024-05-20:
Yes, you can power the board through the VIN pin, on some boards it's connected the same 5V line as the USB VBUS pin, on other boards it's connected through an additional diode. Just make sure you don't power the board through VIN and through the USB connector (from PC) both at once.

Vortecsz, 2024-05-20:
Perfect all working i found some problem; the qualitify of the music is not the best, the sound has some distortion in the background, i'll attach a video below. Sometimes the esp(i think) crashes and the bluetooth is disconnected, and the can also, after some seconds it recconect automatically. Did your setup also have some distortion noise with the aux? I have an aux bluetooth buyed from amazon and it doesn't has background noise

https://github.com/PNKP237/EHU32/assets/61433353/eaeeabb6-ca29-4399-9362-cfa626972faa

PNKP237, 2024-05-20:
I found that the ESP is prone to crashing when the 5V source is not very good. Did you try with additional electrolytic capacitors soldered to the ESP32 board, between VIN and GND? ESP32 is way more power hungry compared to those bluetooth audio boards found everywhere, which is what I used before. My own setup uses a linear 5V regulator (7805) taking power from the radio itself with many additional capacitors on the 12V and 5V lines.
Distorted sounds like in your video do happen in case the I2S driver has no new data to process (it loops the last received music sample, often when bluetooth gets disconnected or gets out of range) or during a crash, but it's never that long, maybe 1 sec max.
I have honestly not used the external aux much since I bought a CD70 from scrapyard and it has no aux circuitry (therefore my EHU32 is soldered inside the radio directly to aux). Although I did test it briefly with the original CD30MP3 (EHU32 connected to external aux and OBDII port) my car came with and I only had some slight noise from unsmoothed DC-DC converters when I tried to power it with the worst quality step down board I could source.

Vortecsz, 2024-05-20:
I'm currently taking the power from an 12v car charger, so i don't think the power sucks. I can use the 7805 as you said? Right now i'm testing all the setup, i'm planning to take the power from the radio as well. For now i can use the 7805 after my car charger (5v)? That one? ; [https://it.aliexpress.com/item/1005006104819794.html?spm=a2g0o.productlist.main.1.2c773abdQfLXOu&algo_pvid=74d1cfe5-e147-4ab0-ac98-43b719560ff8&aem_p4p_detail=20240520140522282305984495550010223217&algo_exp_id=74d1cfe5-e147-4ab0-ac98-43b719560ff8-0&pdp_npi=4%40dis%21EUR%212.19%211.09%21%21%212.33%211.16%21%40211b80d117162391219637920e4d11%2112000035766113066%21sea%21IT%212067665846%21&curPageLogUid=CJeCkjfMOKWo&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=20240520140522282305984495550010223217_1](url)

PNKP237, 2024-05-21:
I should probably clarify in the readme that you should not use the phone's volume control and leave it on max since volume control on I2S results in the audio quality getting worse (the lower the volume, the less bits per sample of audio there is).
Only use vehicle's volume control and optionally you can go into CD40 settings and modify the "Aux in volume" to a value of -5 to lower the noise floor.

The best setting for the CD30 family (so CD40USB as well) would be to use the 7805 inside the radio, it makes the experience more seamless since it powers on/off with the radio itself by using the switched 12V or 8V voltage rail inside the radio unit, but an ESP32 module with an antenna is highly recommended because the radio cage acts like a shield. I actually did test it inside the CD30MP3 as well and the method of connecting it inside should work for CD40 as well.
As for caps I think I've got 470uF + 100nF on the input of the 7805 and 1000uF + 100nF on the output, additionaly there is a 220uF cap soldered directly to the PCM5102 board. MS-CAN is hooked up directly to the legs of TJA1041 CAN transceiver, audio is soldered to the internal AUX vias. Power can be taken either from switched 12V (on the tab of this large SOT-223 package on the right) or from a 8V rail powering the CD drive.
Here are the photos of the mod:
[image]
Internal Aux vias:
[image]
white is left channel, black is audio ground, red is right channel

Vortecsz, 2024-05-22:
Thanks so much, I'll follow your "schematic" as soon as I can. Btw,my radio hasn't worked in FM mode for a year now, it remains in search and no noise is emitted, do you have any idea what it could be?  Has it ever happened to you? Sorry for the OT

PNKP237, 2024-05-22:
I had the radio in my old insignia do it when it got very cold outside (like -10c) but as for cd30 and cd70 there was no issues with the FM reception. My guess is cracked solder joints or a fault with an antenna, next time you will be removing the radio unit from the vehicle try fitting a long wire antenna in place of the antenna connector while leaving the other connector in place so you can test with a substitute antenna.

KingSilverHaze, 2024-09-29:
I think this project is great. I bought the ESP 32 WROOM 32U and flashed the software v0.9.2. Unfortunately I can't get a Bluetooht connection. The name is not displayed with the Bluetooht devices. I then flashed ESP32 Audio Stereo for testing. The Bluetooht name ESP32 ADP is displayed, but no connection is established. What is the error?

PNKP237, 2024-09-29:
> The name is not displayed with the Bluetooht devices.

With release versions, the bluetooth connection is only available when the board is connected to the vehicle's MS-CAN bus (pins 3 and 11) using a CAN transceiver module (such as MCP2551, SN65HVD230 or TJA105x). Once both the radio and the car's display are on then the bluetooth part is initialized.

As for ESP32_Audio_Stereo not working it could be that the board is reset when trying to connect, I'll investigate with my boards.

KingSilverHaze, 2024-09-30:
Thank you for your answer. I have installed it in the CD30MP3 and it works perfectly. The car is a Corsa D. Only the speed is displayed too high. But it is important to me that it works with the Bluetooth stream. Hope this project continues, it's a great alternative for an older vehicle.

Thanks for your work and the know-how.

PNKP237, 2024-09-30:
Glad it works - thanks for the feedback.
I'm working on an update and I'd like to fix the speed readout - if you can, please tell me what speed value is displayed on the screen when you're driving, for example: 20 km/h, 50 km/h, 70 km/h (shown on the speedometer). This way I'd be able to correct the conversion.
Also provide me with an engine code of your vehicle for clarity.

KingSilverHaze, 2024-10-02:
Today I looked at how the values are displayed. 
It is the A14XER engine.

The RPM shows 3500 and the EHU32 shows 1400 RPM.
When idling, the EHU32 shows 3000 RPM.
When I drive at 130, the EHU32 shows 60 - 65.

PNKP237, 2024-10-02:
Thanks for taking the time note down the data. Earlier, you said that the speed was displayed too high, but at 130 km/h it was showing half the actual vehicle speed, is that correct?

NME157, 2024-10-20:
Hello,

Thank You for your great input! I was looking for something similar and what I found is even better. Just wanted to ask if it is possible to connect mic that esp32 would act as handsfree device? What code changes should be made? Btw,will it show on the screen the name of the caller as it is now or code changes should also be made?

PNKP237, 2024-10-20:
>Just wanted to ask if it is possible to connect mic that esp32 would act as handsfree device?

There's currently no support for Bluetooth HFP profile under arduino from what I was able to gather, but if I manage to move away from the arduino core I'll try to implement basic HFP where you can hear the caller through the car's speakers but it will still use your smartphone's microphone.
Showing caller ID and stuff like that should be easy if I can manage the first part.

NME157, 2024-10-20:
> There's currently no support for Bluetooth HFP profile under arduino from what I was able to gather, but if I manage to move away from the arduino core I'll try to implement basic HFP where you can [...]

Thank You very much for quick reply!

NME157, 2024-10-21:
By the way,one more question - song name and artist is only displayed of music is played by phone player or does it display info even of music is played from Spotify?

PNKP237, 2024-10-21:
Depends on the application playing the music, but in my experience it works with all default music players, YouTube app and Spotify. Works with Spotify on iPhone as well.

PNKP237, 2024-12-08:
Hello guys, back to the topic on hand - engine coolant temperature problem
Here's what I got working, coolant temp being displayed on top of the audio metadata:
[image]

This window can be updated separately from the audio stuff which makes it easier. It also works without electronic climate control, it does not even have to be enabled in variant programming (no need for opcom, just plug and play), nor does it influence the temperature set on the AC panel, so no need to worry that the car will start blasting you with 90C+ air.
There's just one downside, if the temperature exceeds 99 deg C, the degree symbol won't be shown, but still I'll try to implement it in v0.9.4.

Tomi8609, 2024-12-10:
Hello

How often does this window pop up or is it there all the time?
Is it possible to display the battery voltage in the same way?

PNKP237, 2024-12-10:
It can be there all the time, constantly updating or be shown for just 5 seconds (just like AC does). 
As for battery voltage, this window only fits 3 characters so no decimal point, seeing a battery voltage like "12V" or "13V" is a little pointless.

Tomi8609, 2024-12-10:
I understand

Thank you

Tomi8609, 2024-12-10:
I think it's enough if it's only visible for 5 seconds

PNKP237, 2025-01-11:
I tried this but for some reason my GID (from a 05 vectra) wouldn't respect the formatting sequence. Not sure what's the issue here but I'll try again some other time.

devilsoul13, 2025-01-13:
hello I love this project it's been a long time that I want to interact with the canbus of my astra h from my side I want to display the information of my cid on a tablet or a phone can you help me out? what would I need as a canbus module to read the information? what is the cabus line for this information (ms can lcan h can) for the development part of the application no problem but I do not know what hardware to use to communicate with the canbus. thanks

PNKP237, 2025-01-13:
The display is connected to both LS-CAN (GMLAN) and MS-CAN so if you wanted to replicate its functionality you'd have to read data from both buses. You could use bluetooth serial port to transmit CAN data to your tablet.
ESP32-C6 is pretty good if you want to interface with two CAN instances. There are cheap C6 development boards you can use with some CAN transceiver modules like MCP2551 or TJA1051. Classic ESP32 (like the one EHU32 runs on) only has one CAN peripheral. In terms of hardware, look at the .pdf "schematic" in the root of this repository.
STM32F4 boards are also pretty good if you don't need bluetooth/wifi, some of them even have 3 CAN peripherals.

devilsoul13, 2025-01-13:
first of all thank you for the speed of your response.
in my country (Tunisia) there are these cards 
can you tell me which is the best because there is a big difference in price.

https://www.celectronix.com/cartes-et-kits-stm32/2810-stm32f4-discovery-stm32f407g-disc1.html
https://2btrading.tn/accueil/8778-carte-de-develeoppement-esp32-c6-devkitc-1-8mb-spi-flash.html

with which canbus I have to work MCP2551 or TJA1051?

PNKP237, 2025-01-13:
> https://2btrading.tn/accueil/8778-carte-de-develeoppement-esp32-c6-devkitc-1-8mb-spi-flash.html
Get the ESP32-C6 then, it's easier to work with under arduino or ESP-IDF compared to STM32. Setting up bit timing is also easier with ESP32. The ESP TWAI driver isn't perfect but works well enough if you're not transmitting much.

> with which canbus I have to work MCP2551 or TJA1051?
any CAN transceiver module will work for MS-CAN or HS-CAN (for example SN65HVD230, TJA1051, TJA1050) just get whatever is cheaper for you. Remember that one board can only handle one CAN bus.
as for interfacting with LS-CAN it's more complicated because it's a single wire CAN line, for reading data you could just use a transistor to invert the signal but for transmission/receive combined an actual SWCAN transceiver is pretty much required, I used MIKROE-4225 because it's pretty much the only SWCAN module available locally

devilsoul13, 2025-01-13:
I would like the most efficient and easiest solution especially for assembly to avoid soldering errors and other

devilsoul13, 2025-01-14:
what do you think about can shield and arduino ? beacause its simple to connect (plug and play)

PNKP237, 2025-01-15:
Classic arduino (atmega328 I guess?) is slow and canbus shield is an external CAN peripheral (MCP2515) and as such it'll be even slower (runs over SPI). Also MCP2515 (not to be confused with CAN transceiver MCP2551) can't run on MS-CAN by default because the timing is wrong (can't set 95kbit unless the clock crystal is changed). It can only support only a single CAN bus at a time.
Just buy jumper wires to connect the boards

devilsoul13, 2025-01-16:
ok THANKS .
so the best configuration for my project would be :
https://2btrading.tn/accueil/8778-carte-de-develeoppement-esp32-c6-devkitc-1-8mb-spi-flash.html
and x 2 https://2btrading.tn/accueil/624-module-sn65hvd230-can-board.html
(2 canbus sn65 one for ms and one for ls can)?

[29 more replies]

closed 2025-04-11 by PNKP237