PNKP237/EHU32 #4: Core dump check failed

kybok12321, 2024-08-03:
Hello, please tell me what could be the reason? I installed the firmware using ProgramESP32.bat and after rebooting the board in SerialMonitor it gives the following error:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (587) esp_core_dump_fl���h: Core dump data check failed:
Calculated checksum='de456912'
Image checksum='ffffffff'

PNKP237, 2024-08-04:
I don't think that's a problem, from what I can gather it's normal since there is no core dump on your ESP32 (the program hasn't ever crashed, therefore there's no valid crash data in there). Have you used your ESP32 board for some other program before?
Does the board reboot continuously?

kybok12321, 2024-08-07:
Thank you very much for your reply and for your excellent work on this program. After this error nothing happens, the board does not reboot. After pressing the reset button I noticed that the check engine briefly lights up and goes out, but nothing else happens. Unfortunately I didn't have the mcp2551 chip board available so I changed to a TJA1050 I2C chip, could this be the problem? Also I don't see bluetooth to connect to. The Esp32 board is in good working order, other programs with bluetooth operation were running on it. Can you please tell me, what should be the behavior of the tape recorder or the car when starting the board? I also tried to compile in Arduino IDE, but I get a lot of errors, one of them is in line 13 of A2DP.ino file : 'title_buffer' was not declared in this scope

PNKP237, 2024-08-07:
Can you send a link to the exact type of ESP32 board you have? A detailed photo is fine too.
I tried all the boards I have around and couldn't replicate the issue, even tried fresh ESP32-WROOM modules straight from the tape and they seem fine. You can also show me the full output after running ProgramESP32.bat.

TJA1050 should work, but I have not tested it myself. Bluetooth only starts after ESP32 receives a message from the radio to the display, then prints a message "Bluetooth on, waiting for connection" (ignition is not necessary), afterwards the bluetooth is available to connect to. What is your vehicle, display and radio unit?

As for compilation, I think you're using an older version (1.x.x) of arduino IDE, update to something newer than 2.x and it should compile fine. While compilation will succeed, there's still an issue with the newest release of ESP32 arduino board files, CAN seems to crash very often with these, also the resulting file will not fit in memory unless you modify the ArduinoOTA library. I'd suggest downgrading to 2.0.17 version of the ESP32 board files for everything to work properly (don't forget to do what I have described in [this section of readme](https://github.com/PNKP237/EHU32?tab=readme-ov-file#compilation-notes) after downgrading/upgrading the board files).

kybok12321, 2024-08-07:
[image]

kybok12321, 2024-08-07:
[image]

PNKP237, 2024-08-07:
I've got the same board in my test setup and it works well, same serial chip and all.
Here are the settings.
[image]

kybok12321, 2024-08-07:
Can you please tell me the versions of your libraries to compile?

PNKP237, 2024-08-07:
ESP32-A2DP v1.8.3
arduino-audio-tools v0.9.8
esp32 board files v2.0.17

Also, I've got one thing for you to try, maybe erasing the entire flash before programming will work in your case. Use the attached batch file to program your board.
[ProgramESP32_erase.zip](https://github.com/user-attachments/files/16532697/ProgramESP32_erase.zip)

kybok12321, 2024-08-08:
Super, thanks a lot, everything compiled perfectly now. However, the problem is still the same, no response. I assume I have wrong soldering, can you tell me if CAN High from TJA1051 is connected to Can high in OBD2 pad (pin 6) and CAN Low is connected to Can Low in OBD2 (pin 14)? When I press RESET on the board, what happens in the video, is it supposed to happen?

https://github.com/user-attachments/assets/5603b010-11f0-464a-976c-0c9c7da23fac

[image]

kybok12321, 2024-08-08:
That's what it was all about! Thank you so much, everything works great. I didn't know Opel had a separate CAN line, I was using the standard obd2 pinout.

closed 2024-08-08 by kybok12321