PNKP237/EHU32 #6: Stuttering audio

xddxd, 2025-02-03:
Hi,

So I’m having the same issue as the person on #1 , the audio on an iPhone 15 Pro running iOS 18.3 is crackling, stuttering and jumping. Meanwhile a Nokia 8 on Android 12 (PixelExperience 12.1) is playing almost flawlessly (have to keep audio down on the phone down otherwise clipping happens). I don’t know what codec or sampling rate the iPhone is using, but the Android connects using SBC and 48KHz.

PNKP237, 2025-02-03:
I used an iPhone 15 (non-Pro) in my vehicle with EHU32 and tested it extensively since I knew that the underlying library for handling A2DP audio has had problems with iOS devices in the past but never had any issues. Found something else here with a suggested fix: [SOLVED Poor quality audio from IOS devices. #582](https://github.com/pschatzmann/ESP32-A2DP/discussions/582)
Certain ESP32 boards seem to be an issue here.

Made a build of EHU32 v0.9.3 with the data out pin changed for you to test (also compiled with updated libraries). Can't really test it on my hardware.
Before you flash your board please remember to connect the DIN pin on PCM5102 to D19 (instead of D22).

[EHU32_093_DOUT_19.zip](https://github.com/user-attachments/files/18647029/EHU32_093_DOUT_19.zip)

Also please describe your board, is it the type-C version? What's written on the metal shielding? What USB-UART chip is on the board, is it the CP2102 or WCH something or other?

xddxd, 2025-02-03:
I’ll try out this version and report back, I can’t tell what is written anymore because the esp32 is glued into a box and is upside down.  I do remember the UART chip being an obvious clone though. Safe to say the rest of the board is likely chinesium.

I have to say though, amazing work, I hate aftermarket radios and this is just perfect, almost jumped out of my chair when I found this.

PNKP237, 2025-02-03:
Got myself an older iPhone for testing but I would appreciate it very much if you could take the time to describe/photograph the board in detail, I've got many variants in my collection but testing each one of them is not feasible, so I want to narrow it down to the problematic ones and work out a fix (mainly for other people attempting to put this together).

I'm in similar boat here since this project is partially a product of my disdain for aftermarket headunits which look out of place and don't integrate with the car like the stock headunits do. Also, how often do you see Aux displayed? I'm still in the process of refining the display blocking/overwriting logic but you're the first person with anything other than CD30 sharing feedback so I'd like to ask. I do most of the development on CD30MP3 with GID and EHU32 works seamlessly with that but each time I test new code with CD70 there's always something that doesn't work exactly right, still getting Aux to show up here and there (probably RDS and navi data getting through and messing with my logic). Not to mention that debugging is difficult when you've got to focus on driving. Hope you understand my frustration here.

xddxd, 2025-02-03:
Aux does randomly pop up yeah, I haven’t noticed a pattern though. Sometimes it doesn’t pop up for a long time, sometimes the text goes blank for half a second and goes back to normal. 
There’s some character limit too it seems, if the first and second lines are really long the 3rd line gets cut off and ends with a "ÿ”. 
Also voltage display is incorrect, shows anywhere between 18-24V, sometimes 4V and doesn’t react to actual voltage changes so it probably is reading bogus data. Likely a difference between Astra H/Zafira B and Vectra C/Signum ECC units.

Aside from that, having steering wheel control on bluetooth is amazing, I should be able to provide you with pictures of the ESP tomorrow. I also tried an iPhone 11 but it exhibited the same audio issue. Looked 100% like a Bluetooth issue, I muted the iPhone but audio was still playing.

xddxd, 2025-02-04:
Here are some board pictures.

[image]
[image]
[image]

PNKP237, 2025-02-04:
> There’s some character limit too it seems, if the first and second lines are really long the 3rd line gets cut off and ends with a "ÿ”. 

I might try to extend the buffer data to support up to 512 chars in some later versions. As for ÿ showing up that's rather strange, unfortunately I don't have a CID on hand to investigate.

>Also voltage display is incorrect, shows anywhere between 18-24V, sometimes 4V and doesn’t react to actual voltage changes so it probably is reading bogus data.

In the next release I'll be moving away from reading data from ECC and read all the data (including voltage) from the display. Hopefully that will work better across different models

>I muted the iPhone but audio was still playing.

That's actually proper behavior since I have forced the audio receiver to NOT apply any volume control and iOS respects that, unfortunately android's behavior varies with manufacturer and once it detects no volume control it applies the volume correction locally before sending the data off to the bluetooth receiver. Not only does that reduce the audio quality (lower dynamic range) but also can result in clipping (as you have observed with your Nokia smartphone). Not much to be done about that other than setting the volume to 50% and hope no correction is applied.

xddxd, 2025-02-04:
Images of the “ÿ” bug. 

[image]
[image]

PNKP237, 2025-02-04:
Thanks for the photos, tried the same album/title/artist (from apple music) with GID but it works fine here but gets cut off earlier. Can you tell me what's the GM part number for that display? I vaguely remember that some vectras and signums had an early CID model (siemens vdo too) that only has two lines of data on the vehicle tab (the rightmost one), is it that one?
Have you been able to check whether moving the I2S data out pin to D19 changed anything in terms of playback?

xddxd, 2025-02-04:
Hi, swapped the pin and flashed the software, no change in iOS but Android now plays fine upto like 80% volume. Display is 24461297.

PNKP237, 2025-02-04:
Right, thanks for trying it out and all the feedback, I'll do some testing with the iPhone and different ESP32 boards over the upcoming weekend. Also, if you have some spare time and any other ESP32 boards at your disposal you might wanna try testing with them.

xddxd, 2025-02-06:
I do have more boards, unfortunately they’re from the same batch and identical to the one I’m using, doubt they’re going to make a difference.

PNKP237, 2025-02-09:
Do you happen to have an aliexpress listing for that board? I have tested 5 different ESP32 boards today and couldn't replicate the issue, audio is fine with an iPhone X running iOS 16.7.10 playing music from Spotify.

<details><summary>Click here for a photo of the tested boards</summary>
<p>

[image]

</p>
</details> 

One of the boards is from WeAct, others are random (ESP-WROOM-32, ESP32-WROOM-32D, ESP32-WROOM-32UE, ESP32-WROOM-32U). Note that your ESP32-WROOM-32D module visually differs from mine, it's likely that one of them is a clone (or both are clones).

marcev31, 2025-04-19:
Hi, I have the same problem as xddxd, the sound stutters on my iphone 13 pro max, I have an android tablet and it plays the sound normally, but the samsung s22 sound stutters. I updated the 0.9.4 firmware today, the problem remains.

PNKP237, 2025-04-20:
I really did try all the boards in my possession but couldn't reproduce the issue, even with iPhone 15 running the latest iOS (as of 2 weeks ago). Is it possible for you to get a different ESP32 board? Personally I'd recommend **WeAct Studio ESP32 D0WD v3** if you're able to get one for testing. All the other no-name boards seem really hit-or-miss with quality.

PNKP237, 2025-04-26:
Finally got some ESP32 modules where audio is stuttering on iPhones but works well with my android device. It honestly seems like some weird hardware issue since I can't find any workarounds for the issue (at this moment).
All the offending modules are designated as **ESP32-WROOM-32E** (certainly clones) and use BoyaMicro flash chip underneath the RF shield. Later this upcoming week I'll try to transplant this memory onto another ESP32 module to see if that makes a difference then we'll know for certain.
Here's a photo of one of the ESP32s exhibiting the issue, so you'll at least know what kind of a module to avoid.
[image]

kotucocuk, 2025-09-14:
The module in the photo has not caused me any issues and works flawlessly with both iPhone and Android.

[image]

maks013-dev, 2025-12-29:
> Finally got some ESP32 modules where audio is stuttering on iPhones but works well with my android device. It honestly seems like some weird hardware issue since I can't find any workarounds for the [...]

@PNKP237 friend, tell me, I also built a similar module for myself, but my songs stutter on my iPhone 16e. I removed the hardware screen from the ESP32, there is an SPI like the one in your photo. Transferring the SPI to another ESP32 helped me figure out the problem?

maks013-dev, 2026-01-03:
> It didn't help in any way for me, did it help in your case? I think it is a problem with the ESP32 chip itself. Either way, I had way more luck with boards which have an antenna connector instead of a PCB antenna (except for the ESP-32S modules, these are garbage).

I bought an original ESP32 Wroom module and re-soldered it onto the board, and the problem went away.[image][image]

KetSchnaider, 2026-01-06:
> I bought an original ESP32 Wroom module and re-soldered it onto the board, and the problem went [...]

Do you have a link to where you bought a good module? I also came across a fake and have the same sound problem as you.

Technolog83, 2026-03-06:
> У меня три типа плат, которые я купил на Aliexpress. Я снял защитную пленку и сфотографировал. На данный момент я еще не проверил, какая из них имеет проблему с iPhone. Как только смогу проверить, сообщу вам.
> 
> [image] [image] [image]

Each of these boards has a problem with the iPhone.

patrikmckane, 2026-10-17:
This one also has stuttering on iPhone issues

[image]
[image]