PNKP237/EHU32 #8: Increasing the range of the Bluetooth signal

Technolog83, 2025-04-30:
I tried installing a simple Bluetooth 5.0 into the CD70 and it was inside the metal radio case. The Bluetooth signal was excellent.

[image]

 I am currently using ESP32 and I have a weak signal. If I use a board with a built-in antenna and place the ESP32 inside the radio (near the bottom CD drive), the signal is almost non-existent. So I use the ESP32 board to connect an external antenna of this type:

[image]

The antenna is glued to the inside of the radio's front panel, next to the volume control. The Bluetooth signal is good, but not strong enough. Sometimes the connection is lost if the phone is in the back seat.

How can I increase the signal? What antenna do you use and where do you place it?

[image]

PNKP237, 2025-04-30:
It all depends on the quality of the ESP32 module used (not the board exactly, but the module with the RF shield itself) - I use a [similar antenna](https://www.aliexpress.com/item/1005001335343983.html) also placed in the same spot as you do and I get solid connection even 5 meters away from my car. You could either try a laptop wifi antenna (if you have some scrap laptops handy) or a different ESP32 module altogether.

It's very unfortunate that it's difficult to get quality ESP32 modules, some are made cheap missing antenna filter components while others produce issues like stuttering music with iOS. The [module I use](https://github.com/PNKP237/EHU32/issues/3#issuecomment-2121866276) seems to be original (based on the Espressif logo) but as with all that comes from aliexpress/alibaba you never really know exactly what's original.

StarX-2401, 2025-05-03:
I use an old bluetooth antenna from an old VAIO laptop which is threaded through one of the holes on the back of the unit and is just resting behind it, it works good for my Samsung A52s phone but not great for some phones such as my Galaxy S3. An IPEX to SMA antenna should work good, I've bought one to upgrade it from the laptop antenna, they are very cheap online (I paid £4 for mine off ebay).

closed 2025-09-05 by PNKP237