PNKP237/EHU32 #15: BID screen problem

kotucocuk, 2025-09-02:
Hi, thank you for sharing such a wonderful project! However, I’ve noticed that the text on the screen doesn’t quite fit properly. Maybe write "BT connected" "BT disconnected" "Coolant : 70°" Here are a few examples below:

[image]
[image]
[image]

PNKP237, 2025-09-03:
Please test this one, just replace EHU32.ino.bin file with the one from the attached zip and reflash the board. I tried to optimize the strings to fit BID better, this will get implemented in the release version when I figure out a way to request the display info.

[EHU32_095_BID_optimized.zip](https://github.com/user-attachments/files/22124719/EHU32_095_BID_optimized.zip)

kotucocuk, 2025-09-03:
> Please test this one, just replace EHU32.ino.bin file with the one from the attached zip and reflash the board. I tried to optimize the strings to fit BID better, this will get implemented in the [...]

Thank you for your interest. I will try tomorrow evening. Have a good evening.

kotucocuk, 2025-09-04:
Hi, i think bt messages enough but coolant still same

[image]
[image]
[image]

PNKP237, 2025-09-04:
Apologies, I only changed the string for data sourced from electronic climate, which your car lacks. Now it's fixed for the display as well, here you go:

[EHU32_095_BID_optimized_v2.zip](https://github.com/user-attachments/files/22142606/EHU32_095_BID_optimized_v2.zip)

PNKP237, 2025-09-04:
Is the coolant temp accurate compared to the one you see in the testmode (settings held for 5s, then pressing BC until coolant is shown)?
I'm asking because you're the first person with BID in their vehicle to use EHU32, I'd like to make sure it reports the temperature properly. I've got a BID but I only test it on my desktop setup at home, hence I only ever see the coolant value of 73 C.

closed 2025-09-04 by PNKP237

kotucocuk, 2025-09-04:
> Is the coolant temp accurate compared to the one you see in the testmode (settings held for 5s, then pressing BC until coolant is shown)? I'm asking because you're the first person with BID in their [...]

I haven’t checked the value by holding the BC button yet, but I’ll do it this weekend(saturday) and get back to you.

PNKP237, 2025-09-04:
Then I'm extremely happy that even a geography teacher was able to make use of my work and put it to use in their car :)

kotucocuk, 2025-09-05:
Hi, my trip was cancelled, so I was able to test it today. Based on the test, I believe it might not be showing the correct result.

[image]
[image]
[image]

PNKP237, 2025-09-05:
Okay, that complicates things, but it's good to know that BID does not report the temperature the same way GID and CID does. I'll try to find the correct value later next week since I have to set up BID with CD30MP3 to test it thoroughly (my usual testing setup is a GID with CD70navi).

PNKP237, 2025-09-05:
No need for that, I have my own custom tools to interact with the CAN bus directly so I can issue custom commands to the display.
Afterwards I'll prepare a special build of EHU32 for you to test in your vehicle. May I also know the year of your Astra and the engine code?

kotucocuk, 2025-09-05:
Okey, Opel astra H 2013 hatchbag A16XER

> No need for that, I have my own custom tools to interact with the CAN bus directly so I can issue custom commands to the display. Afterwards I'll prepare a special build of EHU32 for you to test in your vehicle. May I also know the year of your Astra and the engine code?

kotucocuk, 2025-09-14:
Hi, i completed the project and installed it in the car. The issue was resolved after starting the car, and everything is now displaying correctly. Thank you for looking into it. By the way, would it also be possible to change the music using the arrow buttons on the head unit?

[image]
[image]

PNKP237, 2025-09-14:
Hi, thanks for getting back to me, glad it works after all. Haven't had the time to set up BID with cd30mp3 this week.
Thank you for sharing the photo of the module as well.

As for the buttons, you mean the left and right buttons on CD30MP3, I think it could be done. I'll put it on my to-do list.
I have previously tested the 4-way button (present on older CD30MP3 and all CD70Navi) but it does not send any messages on the CAN network - it's strictly for Navi/radio use and there's no way to read the keypresses.

kotucocuk, 2025-09-22:
Maybe it can display on the EHU32/WELLCOME/OPEL/Bluetooh On. The OTA process takes about 90–120 seconds in total, and the progress from around 10% to 90% seems to take longer than the rest.

> Thanks. I've ran into the issue several times but always forgotten to fix it - the display is not exactly "frozen", EHU32 just keeps retransmitting the last message to overwrite Aux. I'll think of some placeholder text to put there when switching from vehicle data to audio metadata. Out of curiosity, how long did the OTA update take? About one minute or longer?

kotucocuk, 2025-10-05:
Hello,

On Friday, I took an uninterrupted drive of about 4 hours, which gave me a chance to examine the EHU32 in more detail, and I encountered a few issues.

First issue: After using the speed display mode for around 30 minutes, the head unit makes a “click” sound and restarts itself. This problem does not occur when displaying the song title or the “coolant” text. My father experienced a similar issue in his car, which was caused by a “Yatour USB AUX” device.

Second issue: After pressing the steering wheel play/pause button repeatedly 4–5 times, the communication between the head unit and the BID screen was lost. On the BID screen, only the “instant consumption” message remained displayed. Turning the head unit off and on again did not solve the issue. When I turned off the car ignition and tried to restart the head unit, it would not turn on. However, after allowing the car to enter sleep mode and then attempting to turn it on again, everything returned to normal operation.

Today, I took another drive of about 2–3 hours without using the play/pause button and without using the mode where I experienced the first issue. The head unit continued to play music without any problems.

PNKP237, 2025-10-07:
Thanks for sharing your findings. I'm thinking about removing the speed readout because it seems to contribute a lot of issues... The displays aren't meant to receive a new message every 300ms, the radio usually resends the data every 5 seconds.
The click and restart happened to me once on a 2h drive but I didn't notice any particular reason it would crash so I couldn't replicate it. I'll add some functionality to force the display reinitialization in next version as a dirty fix when if its ever needed.

PNKP237, 2025-11-12:
Thanks for letting me know, unfortunately, I've yet to find a way to find out how to request the actual GM part number of the headunit, so implementing that might take a longer while

kotucocuk, 2025-11-23:
Hello,
For the BID displays, would it be possible to add engine RPM and battery voltage to the 4 and 5 buttons in the next update? Is this feasible?

PNKP237, 2025-11-23:
I'll consider this but I'm thinking the next version will have split binaries for each type of display.
Also I'm not sure whether I can read RPMs properly in vehicles with no ECC module

PNKP237, 2026-01-18:
Hmm, seems like a quirk of how BID operates with CD30MP3.
Please do the following: hold 9 for about 1 second -> EHU32 will stop writing messages to the display, so "Aux" should show up, with the bluetooth audio still playing. Afterwards, try to adjust the settings in the "Audio" menu. Please let me know if you're able to enter these settings.
If you can adjust the audio settings on BID then, I'll try to think of some workaround to prevent the audio settings being blocked. I have mostly tested the audio menu on GID and it works with blocking just fine. Right now EHU32 scans for text such as "Fader", "Balance", "Bass", "Treble" and "Sound off" (that's what CD30MP3 sends), if it detects any of these it will stop blocking the messages, so audio adjustments should not be blocked.